a406010503/Miui_Camera #1: MIX4，系统是xiaomi.eu每周最新版 没有街拍模式的选项啊

a546109781, 2022-03-05:
其他的音源变焦，手势拍照都有 就是没有街拍。。。

a406010503, 2022-03-05:
由於未在各機型做過測試，新版本可能不再對各機型支援，建議可嘗試安裝早前的全開版本如:[v4.3.002962.0 (220212)](https://github.com/a406010503/Miui_Camera/releases/tag/Miui_Camera_4.3.002962.0)

a546109781, 2022-03-06:
> 

尝试卸载后安装旧版了，也还是没有

a406010503, 2022-03-06:
那麼最大的可能原因就是MIX4硬體上不支援了... 對此我無能為力

a546109781, 2022-03-06:
> 那麼最大的可能原因就是MIX4硬體上不支援了... 對此我無能為力

行吧，感谢解答

a406010503, 2022-10-02:
最近在查找程式碼時有看到小米有對國際版隱藏了街拍模式
因此我們在最近的徠卡版本中有嘗試強制啟用該項功能
如您有需要使用街拍模式的話可以嘗試看看~

a546109781, 2022-10-12:
> 最近在查找程式碼時有看到小米有對國際版隱藏了街拍模式
> 因此我們在最近的徠卡版本中有嘗試強制啟用該項功能
> 如您有需要使用街拍模式的話可以嘗試看看~

谢谢，用了最新版已经可以了！

a546109781, 2022-10-28:
> 最近在查找程式碼時有看到小米有對國際版隱藏了街拍模式 因此我們在最近的徠卡版本中有嘗試強制啟用該項功能 如您有需要使用街拍模式的話可以嘗試看看~

最近用了一段时间徕卡版本，感觉还是有点卡卡的（可能是官方没有适配的原因，难免）
请问可否将最后一个非徕卡版本（3591）也做一个对国际版强制启用街拍的版本，如果可以的话非常感谢！
如果不能实现的话也没关系~

a406010503, 2022-10-28:
3591的版本過於老舊，如果你不介意我可以將目前最新版1110版本的徠卡特性拿掉，這樣就是公版相機介面了。

a546109781, 2022-10-28:
> 3591的版本過於老舊，如果你不介意我可以將目前最新版1110版本的徠卡特性拿掉，這樣就是公版相機介面了。

可以的，非常感谢~

a406010503, 2022-10-29:
https://cloud.cia.yt/s/OPbS1
試用看看... 有問題再回報

a546109781, 2022-10-29:
> https://cloud.cia.yt/s/OPbS1 試用看看... 有問題再回報

完全没问题

closed 2022-11-01 by a406010503

JerryShih, 2022-11-18:
@a406010503 
在release tag找到最後沒有徠卡版本是 https://github.com/a406010503/Miui_Camera/releases/tag/Miui_Camera_4.3.004561.0

這個1110版本 算是最新的無徠卡版本？
https://cloud.cia.yt/s/OPbS1

我只要把 Miui_Camera_4.3.004561.0 裡的apk 換成這個1110版本嗎？
還是會繼續更新無徠卡版本嗎？

a406010503, 2022-11-18:
@JerryShih 目前最新的是1180，加入了徠卡開關，可以自由切換原相機介面。
您只需要下載壓縮檔正常刷入即可，不需要替換。