a406010503/Miui_Camera #2: (10T Pro)Apollo_pro 應能加入480FPS、新版水印及Kino?

Hs0, 2022-04-30:
嘗試用Umi(Mi10)的配置下去套在Apollo_pro(Mi10T Pro)上反編譯後，相機各功能是能夠正常開啟的，但沒有發現跟原有配置之間的區別，兩款機型同樣使用S865+HMX+OV13B10，主攝、廣角及一些介面上不影響其他鏡頭的功能，例如480FPS、新版水印及Kino這些應該是可以開放到相同程度的，不知道是否能夠將一些沒有存在於Apollo_pro上的功能加入？

考量到陸版沒有使用HMX的Apollo_pro，所以原作者那邊沒有加入，如果有機會加入且需要測試的話，我這邊可以協助

a406010503, 2022-04-30:
是的，確實沒有區別，但是考量到該機型的ROM與多個機型共用(K30S、MI10T)，若強行加上Mi 10T Pro的浮水印可能會造成其他機型也一起跟著變成不符合自身機型的水印，因此這裡就沒有加上。
480FPS因為可能會導致部分機型閃退，我會再研究看看是否能夠加入；至於Mi 10T Pro的KinoUI是可以做到的，預計會在下個更新版本當中加入。

a406010503, 2022-04-30:
補充：後來發現到K30S、MI10T和Mi 10T Pro是分開的，或許有辦法可以只為Mi 10T Pro做修改，但是由於該機型僅在海外上市，有可能因修改導致閃退或其他問題，此部分若能夠協助反饋並且能確保穩定那麼也會在下次版本中加入。

Hs0, 2022-04-30:
這部分應該是可以獨立修改的，原版 com.mi.device 裡面的 Apollo_pro 會調用 Apollo 的多數代碼，僅針對HMX部分做調整，因此直接修改 Apollo_pro+Apollo 會產生干擾問題
這邊準備一份將 Apollo_pro 與 Apollo 切割開的smali，目前實測在手邊的10TP上測試正常，通過這種方式修改應該就不會干擾10T及K30S了
[Apollo_pro.zip](https://github.com/a406010503/Miui_Camera/files/8597572/Apollo_pro.zip)

a406010503, 2022-05-01:
試試看，如果有問題再告知。
安裝v3591_0501版本[載點](https://github.com/a406010503/Miui_Camera/releases/download/Miui_Camera_4.3.003591.0_220501/Magisk-MIUI_Camera-v220501-release.zip)

Hs0, 2022-05-01:
目前測試，功能與介面沒有變化，慢動作模式也仍為120FPS

a406010503, 2022-05-01:
嘗試清除相機數據試試...

Hs0, 2022-05-01:
有嘗試過了，另外也移除module後重新安裝過，依舊沒有效果
目前ROM版本為 EU 13.0.4.0 穩定版

a406010503, 2022-05-01:
確認下相機版本是否為3591版，曾有使用者反饋仍維持eu當次版本的相機。
若有使用核心破解也可嘗試直接安裝包內的apk試試。
如果問題依舊，可能需要找找看是什麼原因導致沒有正常套用了...
（本次修改的點主要有新版水印、480fps、KinoUI，這三項都沒變化嗎?）

Hs0, 2022-05-01:
喔喔有了，目前版本號為 4.3.003591.0
剛剛測試過後發現，直接安裝Module沒有效果，但如果核心破解安裝的話三項都能成功啟用且使用正常，那應該是可以正常工作了

a406010503, 2022-05-01:
目前僅對移除模塊有做清理package_cache的指令，安裝腳本應該是沒有對這部分做處理...
也感謝你的協助，我會在留意這點。

Hs0, 2022-05-01:
OK，那這樣應該測試上先確保能夠正常使用了，我先關閉這個issue，謝謝

closed 2022-05-01 by Hs0