a406010503/Miui_Camera #4: 480 fps missing on mi 10t pro

00raq00, 2022-07-14:
Using prerelease v4.3.004700.1 on my mi10t pro i cannot use 480 fps slow motion. This feature is working on latest stable.

Maybe it is possible to enable it on "leica" version 😉

a406010503, 2022-07-14:
Yes, the MI10T series is not compatible with 480Fps slow motion on this version
Since we have decided not to maintain the Leica version of the camera (too many bugs to deal with)
We recommend that you use the official version 4.3.003591.0 (220501) of this project for a better experience.

00raq00, 2022-07-14:
> 

Your link directs to v4.3.004700.1 😉 that i have installed

00raq00, 2022-07-14:
If you will have any experimental version (non public prerelease) i can install and test 😉

a406010503, 2022-07-14:
This version has just tried to modify 480Fps, please confirm whether it can be used normally.

00raq00, 2022-07-14:
[image]
There is missing slow motion 🙄

a406010503, 2022-07-14:
Is this switch normal in the last version?

00raq00, 2022-07-14:
Yeap it is available:

Camera settings -> customise -> feature layout

00raq00, 2022-07-14:
Hmm let me summarize...

Slow motion is missing on 
[v4.3.004700.1 (220711-Preview)](https://github.com/a406010503/Miui_Camera/releases/tag/Miui_Camera_4.3.004700.1_220711)
Pre-release

Slow motion is available but only 120 fps on 
v4.3.004870.0 (220711-Preview)
Pre-release

a406010503, 2022-07-14:
Still no 480Fps? If it still doesn't, then I may need some time to test the problem so as not to inconvenience you.

a406010503, 2022-07-14:
This is the last modification, if it is still not available or there is no slow-motion 480Fps option, it may be that the Leica camera cannot be adapted, or Xiaomi has forced this option to be limited.
The download link is the same, please download and flash again.

a406010503, 2022-08-03:
We took another look at the problem today for the Mi 10T Pro
You should be able to get fixes in this version, so please let me know if it's still buggy.
[Apk-MIUI_Camera-Leica-v220804-release.apk](https://github.com/a406010503/Miui_Camera/releases/download/Miui_Camera_Leica_4.3.004870.0_220801/Apk-MIUI_Camera-Leica-v220804-release.apk)

00raq00, 2022-08-03:
> We took another look at the problem today for the Mi 10T Pro You should be able to get fixes in this version, so please let me know if it's still buggy. [...]

Still doesn't work 🙄

[image]

a406010503, 2022-08-03:
Your question should be the same as reflected in #5
If you already have 480Fps and it's still not available, the Leica may not be able to adapt
I'm sorry about that!
In this release we also fixed the bug that the 10T Pro could not use the magic clone and watermark error
Hope you enjoy this project to the fullest!

a406010503, 2022-08-03:
If there are no other problems, I will close this issue first, and if there are other problems, you can submit feedback again!

a406010503, 2022-08-06:
We've released a version that fixes a number of issues, and you should be able to get those fixes in this release.
I'll merge into the release channel once it's stable.
[v4.3.004870.0 (220806-Leica-Preview)](https://github.com/a406010503/Miui_Camera/releases/tag/Miui_Camera_4.3.004870.0_Pre_220806)

00raq00, 2022-08-06:
Works fine thx

closed 2022-08-06 by 00raq00