a406010503/Miui_Camera #5: 魔法分身閃退&其他問題

Hs0, 2022-08-01:
機型：Mi10TP
系統VER：Xiaomi.EU 13.0.5.0
目前最新版本v4.3.004870.0(220802_Release)在小米10TP上，有遇到魔法分身必FC的問題，進入任何一項功能均會瞬間造成相機FC，原因不明

已知這個問題在v4.3.004700.1 (220711-Preview)以外的各徠卡版本均有此狀況，v4.3.004700.1 (220711-Preview)及非徠卡版在Magisk module 方式安裝時正常使用，但通過核心破解方式安裝也會引發FC

另外機型浮水印部分，似乎是沒有修改到 Apollo.xml 及Apollo_Pro.xml 配置文件的關係，所以在10T/10TP上均仍會顯示Redmi K30SU，似乎是之前分割的配置文件這次沒有成功套用上
還有萬物追焦、運動抓拍、人眼追焦、大師鏡頭等功能在10TP上是無效的，沒意外大概是真的硬體不支援，或許可以直接去除

a406010503, 2022-08-02:
感謝回報，預計下個版本當中修復。
而萬物追焦等功能因為很多人都說想要，所以還是會留著
不想要的話可以選擇關閉

Hs0, 2022-08-02:
OK，忘記問個，480FPS部分還有辦法加回嗎，目前看除了前面反饋的幾個問題以外，其他補全功能都是正常的，所以想再問看看這部分是否能夠加回，感謝

a406010503, 2022-08-02:
應該可以，只是目前暫時找不到，我會再研究看看~

a406010503, 2022-08-03:
[Apk-MIUI_Camera-Leica-v220804-release.apk](https://github.com/a406010503/Miui_Camera/releases/download/Miui_Camera_Leica_4.3.004870.0_220801/Apk-MIUI_Camera-Leica-v220804-release.apk)
試試看是否正常，若它正常將會正式發佈。

Hs0, 2022-08-03:
魔法分身已修復，480FPS部分可看到開關了，但切換後FC
`io.reactivex.exceptions.UndeliverableException: java.lang.UnsatisfiedLinkError: No implementation found for int com.miui.extravideoxmalgo.xiaomiAlgoInterpolator.XiaomiAlgoInterpolatorJNI.MiVideoMeinInit(int, int, int, int) (tried Java_com_miui_extravideoxmalgo_xiaomiAlgoInterpolator_XiaomiAlgoInterpolatorJNI_MiVideoMeinInit and Java_com_miui_extravideoxmalgo_xiaomiAlgoInterpolator_XiaomiAlgoInterpolatorJNI_MiVideoMeinInit__IIII)
	at io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:4)
	at io.reactivex.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:3)
	at io.reactivex.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:1)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:920)
Caused by: java.lang.UnsatisfiedLinkError: No implementation found for int com.miui.extravideoxmalgo.xiaomiAlgoInterpolator.XiaomiAlgoInterpolatorJNI.MiVideoMeinInit(int, int, int, int) (tried Java_com_miui_extravideoxmalgo_xiaomiAlgoInterpolator_XiaomiAlgoInterpolatorJNI_MiVideoMeinInit and Java_com_miui_extravideoxmalgo_xiaomiAlgoInterpolator_XiaomiAlgoInterpolatorJNI_MiVideoMeinInit__IIII)
	at com.miui.extravideoxmalgo.xiaomiAlgoInterpolator.XiaomiAlgoInterpolatorJNI.MiVideoMeinInit(Native Method)
	at com.miui.extravideoxmalgo.xiaomiAlgoInterpolator.XiaomiAlgoInterpolatorJNI.initializeAlgo(XiaomiAlgoInterpolatorJNI.java:1)
	at com.miui.extravideoxmalgo.xiaomiAlgoInterpolator.XiaomiAlgoMediaInterpolator.init(XiaomiAlgoMediaInterpolator.java:6)
	at com.android.camera.module.video.SlowMotionModule.OooOOO(SlowMotionModule.java:2)
	at OooO0O0.OooO0O0.OooO00o.OooooOo.o00O0O00.OooOOO.run(Unknown Source:2)
	at io.reactivex.Scheduler$DisposeTask.run(Scheduler.java:2)
	at io.reactivex.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:2)
	... 6 more
`

a406010503, 2022-08-03:
將480Fps的代碼添加在我的小米10Pro上它正常工作
`/SlowMotionEnum;->OooO0o0`
因此這個問題有可能在Leica版本上無法適配，可能要想想看有沒什麼方式能解決...
我應該會先撤銷這次的慢動作480Fps的改動

a406010503, 2022-08-03:
我做了一點小調整，你再試試看 (下載連結相同)

Hs0, 2022-08-03:
再次測試，依舊有FC狀況發生，這次的調整可能沒有起到作用
不過我這邊剛才也嘗試了邪門歪道的方案，強行套用Thyme的配置後，在10TP上全功能無FC問題，雖然目前尚未能確定是否有潛在問題，但或許可以提供你調整的方向~~

a406010503, 2022-08-03:
由於我身邊的朋友沒有人使用Xiaomi 10T系列，若要對這個問題修復可能需要您提供LINE或TG等聯絡方式以便我方便找出問題...
當然，也有可能無法修復，因為這個項目我都是保持佛系更新的！

Hs0, 2022-08-03:
我這先放個10TP自測可用的smali好了，看看能否先套上去，讓更多人測試是否有問題
至於10T/K30S部分由於手邊沒有裝置，在額外追加功能（如480FPS這種）的部分我可能就無能為力了
[10TP_0804.zip](https://github.com/a406010503/Miui_Camera/files/9253798/10TP_0804.zip)

a406010503, 2022-08-03:
這個是你自己找出來的修復代碼，還是你從別的機型直接整份移植過來的? 因為我怕如果是後者可能會對裝置帶來負面影響。

Hs0, 2022-08-03:
目前是先嘗試整份替換，已知用Thyme替換正常運作，Umi偶發FC，Cmi因硬體差異較大，FC問題尤其嚴重

先加TG好了，再看看如何縮小修改範圍：XXXF_Gi

a406010503, 2022-08-05:
問題大致上解決，近期會發佈一個版本至發佈預覽通道，我先關閉這個議題。

closed 2022-08-05 by a406010503