a406010503/Miui_Camera #16: Dynamic Shots Not Working

abdoelbanaa, 2022-11-12:
Hi brother's the feature Dynamic Shot Note working in some devices please can force showing and working in any device!?

abdoelbanaa, 2022-11-12:
> Hi brother's the feature Dynamic Shot Note working in some devices please can force showing and working in any device!?

Ad my phine is Redmi Note 10

a406010503, 2022-11-12:
你是說動態照片嗎?

abdoelbanaa, 2022-11-12:
> do you mean dynamic photos?

No i mean dynamic shots in photo

a406010503, 2022-11-12:
Dynamic shots are already mandatory on all devices, don't you?

abdoelbanaa, 2022-11-13:
> Dynamic shots should be mandatorily enabled on all devices, don't you have this feature?

Yes, I do not have it and it is present in some other phones. Please solve this problem and review the codes

a406010503, 2022-11-13:
Please check if the latest version has been fixed... [Click](https://github.com/a406010503/Miui_Camera/releases/tag/Miui_Camera_4.5.001180.0_221113)

abdoelbanaa, 2022-11-13:
> Please check if the latest version has been fixed... [Click](https://github.com/a406010503/Miui_Camera/releases/tag/Miui_Camera_4.5.001180.0_221113)

Yeah bro working and the best version everyway

abdoelbanaa, 2022-11-13:
Fixed

closed 2022-11-13 by abdoelbanaa