a406010503/Miui_Camera #17: Camera disappear after last update why?

Pampurios, 2022-11-13:
hi i updated the camera to today's version. after restarting the phone with magisk the camera disappears. i no longer have the icon in the launcher and there is no trace of it. previous version all ok. thank you guys

a406010503, 2022-11-14:
We will try to fix this issue in the next fix version, until then we recommend to watch the modification instructions and install it with Core Patch.

Pampurios, 2022-11-14:
Thank you. I will try the fix. Good job 👍

a406010503, 2022-11-15:
New version 221115 has been released, try to see if it solves.

jfcherng, 2022-11-15:
After installing today's `Magisk-MIUI_Camera-Leica-v221115-release.zip`, my camera disappers as well. Rollback to `Magisk-MIUI_Camera-Leica-v221113-release.zip` and the camera appears again.

rsps1008, 2022-11-16:
My 221115 will disappear…, but 221113 is normal on my mi 12 pro in MIUI11 Eu ROM 13.0.41

jfcherng, 2022-11-16:
Most likely you guys didn't install LSPosed + Core Patch.

rsps1008, 2022-11-17:
Yes, sure I didn't. To be fair, I don't want to install LSPosed if I don't must to, and I can used the module normally before 221113. I think I should stay at version 221113 if I have to install LSPosed in the future, thank you for your reply.

a406010503, 2022-11-17:
Check if version 221117 is available.

a406010503, 2022-11-19:
Since this issue is no longer active, we will close this issue.

The problems mentioned in this issue will continue to be optimized in subsequent versions.

closed 2022-11-19 by a406010503