a406010503/Miui_Camera #20: Port request for K20 Pro

singhh9596, 2022-11-15:
Can you please port miui_camera for k20 Pro (mi 9T pro) as well? Using ANX Camera and it has broken 48mp and potrait mode.

a406010503, 2022-11-15:
Unfortunately, we can't add profiles for each model, Xiaomi adds a lot of obfuscated code which makes it hard to find... We recommend using Telegram's [@sipollo](https://t.me/sipolloMOD) Camera MOD, which has a Xiaomi Leica camera for the K20 Pro based on the 4870 version...

singhh9596, 2022-11-15:
That's unfortunate. And about the module you suggested, I've already tested it and it didn't flashed in Raphael. Thanks by the way

a406010503, 2022-11-15:
You can find it here... [Link](https://t.me/sipolloMOD/308)
As this is out of my scope, I will close this question, if you have any questions you can reopen this

closed 2022-11-15 by a406010503