a406010503/Miui_Camera #21: Track move objects Bug

abdoelbanaa, 2022-11-20:
Hi , The feature Moving Track not working why and after open video mode and tap to Track Move Object And touch screen The Camera Force Close Please Can fix it 

@a406010503 
[image]

a406010503, 2022-11-20:
This feature may not be supported on all devices, which is a problem we tend to encounter with camera patching, which usually happens on phones that have been launched for a long time...

a406010503, 2022-11-22:
This issue will be closed as this issue is no longer active.
If you need anything later, you can reopen this issue or open a new issue.

closed 2022-11-22 by a406010503