a406010503/Miui_Camera #22: Disable Parallel Processing for Portrait mode?

ItzDFPlayer, 2022-11-30:
Is there's any way to disable Parallel Processing only for Portrait mode, I'm making Mi 9T/Pro (davinci/raphael) port, the only thing is broken right now is back Portrait while it's enabled, I'm checking code right now, but maybe you know more

a406010503, 2022-12-03:
Maybe you can turn it off in lab mode...

ItzDFPlayer, 2022-12-03:
I know, it works like that, just curious if there's some check for paralleprocess in portrait to break in my port

a406010503, 2022-12-04:
We don't know how it works for the time being. For K20 Pro, other authors made it, you can refer to the following telegram link: [Click me](https://t.me/sipolloMODgroup/61618)

a406010503, 2022-12-06:
This issue will be closed as this issue is no longer active.
If you need anything later, you can reopen this issue or open a new issue.

closed 2022-12-06 by a406010503