a406010503/Miui_Camera #28: Xiaomi 12S Ultra always on HDR not working

zeno10, 2022-12-10:
The always on HDR toggle works fine, however the behaviour is still auto and is not always active. Using latest release version.

a406010503, 2022-12-10:
Can you be more specific, such as pictures?

zeno10, 2022-12-10:
[image]

When taking pictures using HDR always on the behaviour is exactly the same as HDR auto, the camera decides when HDR is necessary. The toggle works fine as in the picture, but the pictures will always be identical regardless of always on or auto mode HDR.

kazhemons, 2022-12-10:
@zeno10 can you pm me in telegram @vhdcny, to share your photo results