a406010503/Miui_Camera #32: 相机一直出现闪退

WhoIAmilet, 2022-12-15:
[已经删除了对应的Miui_Camera再次重新刷入模块,但相机一直无法打开，一直闪退，已经更新到221213版本，]
[image]
[image]
[image]

a406010503, 2022-12-15:
先確認是否其他模塊造成... 既然已經刪除模塊應該就不是我的問題
另外，請嘗試清除 /data/system/package-cache/ 底下的文件，並重啟您的手機

a406010503, 2022-12-15:
順便問一句，你是否更新到 MIUI 14 了?  目前模塊尚未適配 MIUI 14，請使用核心破解方式安裝相機。

WhoIAmilet, 2022-12-16:
并未更新到miui14，目前版本为miui13.0.24，已经试着清除cache，并未奏效，在酷安平台也看到相机闪退的反馈。最后我选择了清除所有手机数据重新安装，现在已经可以正常使用。
我猜测可能是安装相机补全模块后，在恢复手机备份时安装了系统的相机文件导致。

closed 2022-12-16 by a406010503