a406010503/Miui_Camera #34: Not working on miui 14

DarkHacker2k223, 2022-12-18:
When can this module support MIUI14? Now do you plan to port it on MIUI14?

a406010503, 2022-12-18:
We currently have no plans to support the Magisk module for MIUI 14 as it may have serious issues on some models. It is recommended that you unzip the camera in the installation package and install it directly with Core Patch

DarkHacker2k223, 2022-12-20:
It now works on MIUI 14
[image]
[image]
. but have to bring the first version to flash

closed 2022-12-21 by a406010503