a406010503/Miui_Camera #35: 3.7x zoom button in Mi Note 10/CC9 Pro

rzt12, 2022-12-18:
when the 3.7x zoom button is pressed, the lens used is 2x optical instead of the 3.7x optical in the Mi Note 10/CC9 Pro

a406010503, 2022-12-18:
Since Xiaomi CC9 Pro / Xiaomi Note 10 is about to be unsupported, this issue may not be fixed, and we don't know how to fix this issue, because it works fine on my Xiaomi 10 Pro...

rzt12, 2022-12-18:
well, now I understand what the problem is. thanks for the elaboration of the problem.

closed 2022-12-18 by a406010503

rzt12, 2022-12-18:
sorry in advance, I found 1 Problem again, for the device name on the leica watermark it says redmi note 10, not mi note 10, can it be fixed?

a406010503, 2022-12-18:
@rzt12 This issue will be checked and please verify this issue in the next release

rzt12, 2022-12-18:
[image]

rzt12, 2022-12-18:
well, thank you very much for the response and thank you very much for creating and developing this cool and great mod camera app. I really like and comfortable using it. Hopefully, this project will continue and grow. We will always wait for its development and will also continue to support you and your team 😊