a406010503/Miui_Camera #36: 5x Telephoto Lens in Pro Video Mi Note 10

rzt12, 2022-12-19:
i'm sorry if i ask a lot, i just want to know why the tele 5x lens option that uses a 3.7x optical lens in pro video is no longer available on the mi note 10?. in older or previous versions, Roman Yalta included a 5x Tele option in the video pro, but in later versions until the current version, that Tele 5x option is no longer available.

When I use the 5x tele on Video Pro, I found no issues or bugs. Although it doesn't have an EIS stabilizer, it's not a problem.

sorry if my english is bad. thank you

a406010503, 2022-12-19:
Can you tell me which version was there before? Please provide relevant screenshots...
In Roman's 1110 camera, video cannot be selected tele 5x in professional mode. I don't know why you have it?

romanyalta, 2022-12-19:
> i'm sorry if i ask a lot, i just want to know why the tele 5x lens option that uses a 3.7x optical lens in pro video is no longer available on the mi note 10?. in older or previous versions, Roman [...]

APK https://disk.yandex.ru/d/dGPlNhr6ylUstg
Magisk module https://disk.yandex.ru/d/U1dgffyyJRxTmQ

rzt12, 2022-12-19:
> Can you tell me which version was there before? Please provide relevant screenshots... In Roman's 1110 camera, video cannot be selected tele 5x in professional mode. I don't know why you have it?

[image]

rzt12, 2022-12-19:
> APK https://disk.yandex.ru/d/dGPlNhr6ylUstg Magisk module https://disk.yandex.ru/d/U1dgffyyJRxTmQ
>
> [image]

thank you very much Roman has answered my questions and was kind enough to help me directly by providing me with your great mod app. i really really appreciate it. This is the application or project that I dreamed of and wanted to make since the first time I owned the Mi Note 10, but I don't know and understand where to start. I hope that this great application or project will continue to exist, develop and be useful for all of us. and hopefully more and more connoisseurs of this application or project will voluntarily help donate to you or your team so that they will be even more enthusiastic about developing it

a406010503, 2022-12-19:
I asked him to come here, I think instead of speculating here, let him answer this question...
Since it's not supported on my model (even with Roman's mod installed), I'm probably not going to do this... And mn10 will soon no longer be supported in the new version.

rzt12, 2022-12-20:
> > mohon maaf jika saya banyak bertanya, saya hanya ingin tahu mengapa opsi lensa tele 5x yang menggunakan lensa optik 3,7x dalam video pro tidak lagi tersedia di mi note 10?. di versi lama atau sebelumnya, Roman Yalta menyertakan opsi 5x Tele di video pro, tetapi di versi yang lebih baru hingga versi saat ini, opsi Tele 5x tersebut tidak lagi tersedia.
> > Ketika saya menggunakan tele 5x di Video Pro, saya tidak menemukan masalah atau bug. Meskipun tidak memiliki penstabil EIS, itu tidak masalah.
> > maaf jika bahasa inggris saya buruk.terima kasih
> 
> APK https://disk.yandex.ru/d/dGPlNhr6ylUstg Modul magisk https://disk.yandex.ru/d/U1dgffyyJRxTmQ
> 
> [image]

sorry Roman, I have installed and tried the application you provided, but when I switched the interface from stock cam to Leica, one of the telephoto lenses in the video pro mode disappeared and there was only 1 telephoto lens available. different from when using the stock cam interface(not the leica interface), the two telephoto lenses (2x and 5x) are available in pro video mode

rzt12, 2022-12-20:
> > > i'm sorry if i ask a lot, i just want to know why the tele 5x lens option that uses a 3.7x optical lens in pro video is no longer available on the mi note 10?. in older or previous versions, Roman Yalta included a 5x Tele option in the video pro, but in later versions until the current version, that Tele 5x option is no longer available.
> > > When I use the 5x tele on Video Pro, I found no issues or bugs. Although it doesn't have an EIS stabilizer, it's not a problem.
> > > sorry if my english is bad. thank you
> > 
> > 
> > APK https://disk.yandex.ru/d/dGPlNhr6ylUstg Magisk module https://disk.yandex.ru/d/U1dgffyyJRxTmQ
> > [image]
> 
> thank you very much Mr. Roman has answered my questions and was kind enough to help me directly by providing me with your great mod app. i really really appreciate it. This is the application or project that I dreamed of and wanted to make since the first time I owned the Mi Note 10, but I don't know and understand where to start. I hope that this great application or project will continue to exist, develop and be useful for all of us. and hopefully more and more connoisseurs of this application or project will voluntarily help donate to you or your team so that they will be even more enthusiastic about developing it

> APK https://disk.yandex.ru/d/dGPlNhr6ylUstg Magisk module https://disk.yandex.ru/d/U1dgffyyJRxTmQ
>
> [image]

sorry Roman, I have installed and tried the application you provided, but when I switched the interface from stock cam to Leica, one of the telephoto lenses in the video pro mode disappeared and there was only 1 telephoto lens available. different from when using the stock cam interface(not the leica interface), the two telephoto lenses (2x and 5x) are available in pro video mode

[image]
[image]

romanyalta, 2022-12-20:
reuploaded from the same link

rzt12, 2022-12-20:
> reuploaded from the same link

Thank you very much Roman for the quick response and help. I will install and try again

a406010503, 2022-12-21:
It will be merged into the next 1620 fix based on this change... Maybe I'll release it by the end of the week

rzt12, 2022-12-21:
> It will be merged into the next 1620 fix based on this change... Maybe I'll release it by the end of the week

thank you very much HolyBear team for the response and help. I really really appreciate what you and your team are doing. can't wait for the release of the next fix or optimization before this device is completely deprecated in this project 😭

rzt12, 2022-12-21:
sorry if I put all the issues together here, I don't want everything to separate and get messy.

a little suggestion from me again. As we all know, the 3.7x tele zoom factor button doesn't really work or switch to using the 3.7x optical telephoto lens on the Mi Note 10 in almost every mode. wouldn't it be better that the button just be removed and simply retain 5x only, being (0.6x, 1x, 2x and 5x)?

I'm sorry, I didn't mean to force you to do it, I just wanted to give a little advice to make this application even better and optimal again in Mi Note 10

rzt12, 2022-12-21:
> > HolyBear team
> 
> I have nothing to do with this project.
> 
> > As we all know, the 3.7x tele zoom factor button doesn't really work or switch to using the 3.7x optical telephoto lens on the Mi Note 10 in almost every mode.
> 
> Very strange, it used to work correctly... I don't use mn10 anymore (now I have m11u), but I'll have a chance to check it out.

sorry sir, I thought you were involved in this project because I saw your name included in this project

a406010503, 2022-12-21:
> sorry sir, I thought you were involved in this project because I saw your name included in this project

It is out of respect for the original author that we added his name (Roman), and we have a part of the wisdom of MOD from him, so we would like to thank him...

rzt12, 2022-12-21:
> > Tim Beruang Suci
> 
> Saya tidak ada hubungannya dengan proyek ini.
> 
> > Seperti yang kita semua tahu, tombol faktor zoom tele 3,7x tidak benar-benar berfungsi atau beralih menggunakan lensa telefoto optik 3,7x pada Mi Note 10 di hampir setiap mode.
> 
> Sangat aneh, dulu berfungsi dengan benar ... Saya tidak menggunakan mn10 lagi (sekarang saya punya m11u), tetapi saya akan memiliki kesempatan untuk memeriksanya.

Yes sir, this is strange. even though there used to be no problem, but after a few updates I noticed that when I pressed the zoom factor 3.7x button (at first it was 1x) the lens wouldn't switch to using 3.7x optical, but instead switched to using a 2x optical lens (it felt very unstable because there was no ois). 3.7x optical lens can really be used or moved when we zoom in manually using 2 fingers between 3.7x to 3.8x. tomorrow I will include a sample (switching between lenses) on the mod camera app using screen record

i'm glad you are now using mi 11 ultra, it's a cool smartphone and it could be a great opportunity for you to improve the camera, just like in the old days when you used mi note 10.

btw thank you very much sir, for helping me check and fix almost every issue that I complained about here. the reason I mentioned or involved you from the start is because I feel the application in this project is almost similar to the application you made a long time ago, so I think for sure you will also be part of this project.

rzt12, 2022-12-21:
> 

well, now I understand. thanks for the explanation Mr. HolyBear. I'm sorry for thinking Mr. Roman was participating in your project as well, until I accidentally got him involved here. i'm so sorry

rzt12, 2022-12-21:
> I have nothing to do with this project.
>
>
>
>
> Very strange, it used to work correctly... I don't use mn10 anymore (now I have m11u), but I'll have a chance to check it out.

Yes sir, this is strange. even though there used to be no problem, but after a few updates I noticed that when I pressed the zoom factor 3.7x button (at first it was 1x) the lens wouldn't switch to using 3.7x optical, but instead switched to using a 2x optical lens (it felt very unstable because there was no ois). 3.7x optical lens can really be used or moved when we zoom in manually using 2 fingers between 3.7x to 3.8x. tomorrow I will include a sample (switching between lenses) on the mod camera app using screen record

i'm glad you are now using mi 11 ultra, it's a cool smartphone and it could be a great opportunity for you to improve the camera, just like in the old days when you used mi note 10.

btw thank you very much sir, for helping me check and fix almost every issue that I complained about here. the reason I mentioned or involved you from the start is because I feel the application in this project is almost similar to the application you made a long time ago, so I think for sure you will also be part of this project.

rzt12, 2022-12-21:
> It is out of respect for the original author that we added his name (Roman), and we have a part of the wisdom of MOD from him, so we would like to thank him...

well, now I understand. thanks for the explanation Mr. HolyBear. I'm sorry for thinking Mr. Roman was participating in your project as well, until I accidentally got him involved here. i'm so sorry @romanyalta @a406010503

rzt12, 2022-12-22:
> Yes sir, this is strange. even though there used to be no problem, but after a few updates I noticed that when I pressed the zoom factor 3.7x button (at first it was 1x) the lens wouldn't switch to [...]

https://drive.google.com/file/d/1Hw4RIKp9yam5uln5Sdfpq6eDYPOE2I7t/view?usp=sharing
This is a sample video I recorded from MIUI Cam Mod. The issue was when I switched between lenses by pressing the 3.7x button but the lens didn't switch using 3.7x optical zoom, but instead used a 2x optical lens that was digitally zoomed to 3.7x (that's where the camera looks shaky and unstable).
**this issue occurs in your own and HolyBear's camera mods when used on the Mi Note 10.

sorry if i use google drive link and I'm also sorry to have discussed everything here and involved you Mr. @romanyalta

romanyalta, 2022-12-22:
> 3.7x button

I checked - it works for me correctly (xiaomi.eu 22.7.13). It is very noticeable when the color of the sensor changes during the module change. You may need to try resetting the camera app data.

https://disk.yandex.ru/i/p5pxgAyitJ9xBw

rzt12, 2022-12-22:
> > 3.7x button
> 
> I checked - it works for me correctly (xiaomi.eu 22.7.13). It is very noticeable when the color of the sensor changes during the module change. You may need to try resetting the camera app data.
> 
> https://disk.yandex.ru/i/p5pxgAyitJ9xBw

Thank you Mr Roman for checking and trying it. does when you switch to using the Leica interface, the 3.7x keys also work and there are no problems?. i'm sorry to have troubled you Mr Roman.

I have already seen your screen record and there is no problem on your device. I've followed your advice to clear the camera app including intubating the Magisk module and then trying it again, but it didn't come to fruition. I am currently using xiaomi.eu stable version 13.0.4.0(2022-09-04) base Chinese rom. Previously I was on Weekly 22.5.18. On both ROMs above the result is the same, the 3.7x key does not work on my device. if this issue occurs only on my device, it doesn't matter, at least I already know that this only happens to me

romanyalta, 2022-12-22:
> Leica

https://disk.yandex.ru/i/OvI9e7FJ6WKCaQ

romanyalta, 2022-12-22:
> but to no avail

Perhaps you are using a cover that is in the way?

a406010503, 2022-12-22:
Merge Roman's changes into the [221222](https://github.com/a406010503/Miui_Camera/releases/tag/Miui_Camera_4.5.001620.2_221222) version and check if it works.

closed 2022-12-22 by rzt12

rzt12, 2022-12-22:
> > Leica
> 
> https://disk.yandex.ru/i/OvI9e7FJ6WKCaQ

> > Leica
> 
> https://disk.yandex.ru/i/OvI9e7FJ6WKCaQ

I've seen it and that 3.7x button works really well on your device. but strangely on my device such buttons do not work at all. every time I press the 3.7x button, the lens used is a 2x optical zoom lens which in digital zoom becomes 3.7x. the 3.7x button can actually work when after I press the 3.7x button then I zoom in manually using 2 fingers(the 3.7x telephoto lens works between the 3.7x - 3.8x range, or slightly above 3.7x). I've tried various ways to solve it, starting from reinstalling and replacing the Magisk MIUI cam mod Module but the results remain the same. hm does it have anything to do with the rom I'm using?.

rzt12, 2022-12-22:
> > Leica
> 
> https://disk.yandex.ru/i/OvI9e7FJ6WKCaQ

I've seen it and that 3.7x button works really well on your device. but strangely on my device such buttons do not work at all. every time I press the 3.7x button, the lens used is a 2x optical zoom lens which in digital zoom becomes 3.7x. the 3.7x button can actually work when after I press the 3.7x button then I zoom in manually using 2 fingers(the 3.7x telephoto lens works between the 3.7x - 3.8x range, or slightly above 3.7x). I've tried various ways to solve it, starting from reinstalling and replacing the Magisk MIUI cam mod Module, but the results remain the same.

does this have anything to do with the rom i'm using?

romanyalta, 2022-12-22:
> does this have anything to do with the rom i'm using?

unlikely, but anything is possible

rzt12, 2022-12-22:
> > but to no avail
> 
> Perhaps you are using a cover that is in the way?

No sir, I don't use camera or body protection. the camera is very clean and absolutely not hindered by anything. the photos are also very good, no problem

rzt12, 2022-12-22:
> 

yes, that's right Sir. hm maybe I will try to replace my rom like the rom you are using too. if it doesn't produce results, there may be something strange or wrong with my device and I need to accept the strangeness or lack of it well. at least the 5x zoom button still works very well

rzt12, 2022-12-22:
for everything you do, I am very grateful. thank you for taking so much time to respond and help me, especially in checking and fixing all the issues or problems I have complained about. and thank you, for volunteering to create and develop this great and cool camera app. thanks very much to you, we can all experience a lot of new things without having to leave this device as soon as possible. hopefully someday I or all of us can repay all your kindness and struggle in making this all.
@a406010503 @romanyalta

rzt12, 2022-12-22:
> Merge Roman's changes into the [221222](https://github.com/a406010503/Miui_Camera/releases/tag/Miui_Camera_4.5.001620.2_221222) version and check if it works.

thank you very much Mr. HolyBear for the response, improvements, optimizations and updates. I've updated it and checked it, now everything works very well. except for the 3.7 x button problem that doesn't work on my device, but it doesn't matter, maybe it's just me who experiences it. because on device Mr. Roman everything works very well and I'm sure other users have not encountered this problem either.

rzt12, 2022-12-22:
> > does this have anything to do with the rom i'm using?
> 
> unlikely, but anything is possible

yes, that's right Sir. hm maybe I will try to replace my rom like the rom you are using too. if it doesn't produce results, there may be something strange or wrong with my device and I need to accept the strangeness or lack of it well. at least the 5x zoom button still works very well

rzt12, 2022-12-22:
for everything you do, I am very grateful. thank you for taking so much time to respond and help me, especially in checking and fixing all the issues or problems I have complained about. and thank you, for volunteering to create and develop this great and cool camera app. thanks very much to you, we can all experience a lot of new things without having to leave this device as soon as possible. I'm sorry to have troubled you so much. hopefully someday I or all of us can repay all your kindness and struggle in making this all.

rzt12, 2022-12-22:
for everything you do, I am very grateful. thank you for taking so much time to respond and help me, especially in checking and fixing all the issues or problems I have complained about. and thank you, for volunteering to create and develop this great and cool camera app. thanks very much to you, we can all experience a lot of new things without having to leave this device as soon as possible. I'm sorry to have troubled you so much. hopefully someday I or all of us can repay all your kindness and struggle in making this all.
@a406010503 @romanyalta