a406010503/Miui_Camera #40: MIUI14安裝完後相機不見

Haruyya, 2022-12-21:
我是小米12pro（EU開發版） 安裝完後重啟相機直接不見
解除安裝也沒有恢復 
[image]
[image]

yuejunwudi, 2022-12-22:
我是小米11青春版，刷了v4.5.001620.2（221220）版后相机的app就消失不见了，无法使用相机。
从magisk移除模块后相机app就回来了。magisk版本ed37ddd5-alpha（25205），手机是安卓13，miui14开发版。

a406010503, 2022-12-22:
模塊導致相機丟失問題猜測為更改版本號所致，因為之前也有因為更改版本號導致類似的情形，這是小米所做的嚴格的簽名校驗導致的，我們會在下個版本回復他的版本號，並將修改後的版本號另做一個 APK 上傳他。

a406010503, 2022-12-22:
檢查最新版本 221222 與 221222T 是否正常。

Haruyya, 2022-12-22:
> 檢查最新版本 221222 與 221222T 是否正常。

剛剛解除舊模塊 安裝這個已經正常了不再刪除相機

closed 2022-12-22 by a406010503