a406010503/Miui_Camera #41: Redmi k20 pro issue

hardikdesai97, 2022-12-22:
[image]

Miui Leica camera work in redmi k20 pro but need to disable dual SAT and parallel processing
I hope dev can fix this in next update

a406010503, 2022-12-22:
Since my support for the K20 series is an experiment, we can't guarantee that it will work reliably and I will look into this area.

hardikdesai97, 2022-12-22:
thanks for good response

closed 2022-12-22 by hardikdesai97

a406010503, 2022-12-22:
Check that the latest versions of 221222 are working.

a406010503, 2022-12-23:
The problem is that the feedback we receive is that in most cases you have to turn it off to get it to work, and the K20 Pro has been off the list of supported models for a long time, so we can't fix this problem...