a406010503/Miui_Camera #42: 刷了v4.5.001620.2（221220）版后相机的app就消失不见了，无法使用相机

yuejunwudi, 2022-12-22:
我是小米11青春版，刷了v4.5.001620.2（221220）版后相机的app就消失不见了，无法使用相机。
从magisk移除模块后相机app就回来了。magisk版本ed37ddd5-alpha（25205），手机是安卓13，miui14开发版。
[image]
[image]
[image]

a406010503, 2022-12-22:
模塊導致相機丟失問題猜測為更改版本號所致，因為之前也有因為更改版本號導致類似的情形，這是小米所做的嚴格的簽名校驗導致的，我們會在下個版本回復他的版本號，並將修改後的版本號另做一個 APK 上傳他。

a406010503, 2022-12-22:
檢查最新版本 221222 與 221222T 是否正常。

yuejunwudi, 2022-12-22:
感谢大佬更新，麻烦大佬了。
我刷入了这两个版本Magisk-MIUI_Camera-v221222-Pre-dev-MIUI14.zip和Magisk-MIUI_Camera-Leica-v221222-release-MIUI14.zip还是有问题，刷完以后相机app消失不见了。 另外安装Apk-MIUI_Camera-v221222.apk安装不成功 提示“安装包解析出错”。

closed 2022-12-22 by a406010503