a406010503/Miui_Camera #45: 两个小建议

Chen017, 2022-12-23:
[image]
[image]
第一个延时摄影中 相机设置里的延时摄影与延时摄影模式功能重复 可以删掉相机设置里的延时摄影速度
[image]
[image]

第二个 可以将动态照片移入二次展开栏 这样主界面不会拥挤 与12su一致 我是从thor类里面找到这段代码加入自己机型实现的 具体的跳转我没去找

.method public Oooooo0()[I
    .registers 1

    const/4 p0, 0x2

    new-array p0, p0, [I

    fill-array-data p0, :array_8

    return-object p0

    nop

    :array_8
    .array-data 4
        0xa30ce01
        0xa70d101
    .end array-data
.end method
谢谢

a406010503, 2022-12-23:
感謝建議，但是Google Lens和超級微距也會在上面，這還是不能完全解決問題。

Chen017, 2022-12-23:
ok

closed 2022-12-23 by Chen017