a406010503/Miui_Camera #54: Not working on Miui 14 Android 13 ( MI global version )

xShihaBx, 2023-02-09:


a406010503, 2023-02-10:
The information is not complete, please complete it within the next few days, otherwise I will close this question.

xShihaBx, 2023-02-10:
I am using Miui 14 Android 13 (global rom ) device. Whenever I try to install any of those miui 14 modules my camera app is uninstalled everytime and I can not even install the apk file provided with the module so I had to uninstall the module and after restart my default camera app bring back.

xShihaBx, 2023-02-10:
[image]

xShihaBx, 2023-02-10:
[image]

a406010503, 2023-02-10:
Please read the [modification instructions](https://github.com/a406010503/Miui_Camera/blob/main/Leica_en.md) first, which will show you how to do the Core Patch process, and when you are done you can install the Apk directly.

Magisk module is still unstable on MIUI 14, we recommend you to use Core Patch program to install the camera app.

closed 2023-02-13 by a406010503