a406010503/Miui_Camera #58: watermark Leica

jjoeshua, 2023-04-04:
I install apk version of MIUI camera , and it was after opening Leica, I have xiaomi 12t pro , I turn watermark and looks like this:  xiaomi12t pro | Leica , is it posible to remove that leica?

jjoeshua, 2023-04-04:
supermoon option working? I also didnt find 200mp?

a406010503, 2023-04-04:
I don't know, but there is a theory, 200MP is displayed as "Ultra Pixel" on
your model, and the super moon needs hardware support, and there may be a
chance that it cannot be used.

jjoeshua, 2023-04-04:
I can open it, choose models and that , but wait for night to try to shoot..

a406010503, 2023-04-08:
This issue will be closed as this issue is no longer active.
If you need anything later, you can reopen this issue or open a new issue.

closed 2023-04-08 by a406010503