a406010503/Miui_Camera #61: don't try to install this normal camera app has been deleted

Fukankaya, 2023-06-11:
I started to install this module with magisk, then I left in the middle of the process because my English was not enough and now I saw the camera application as deleted. I downloaded the camera application 5.0 from the internet but it would not open. I deleted this module via magisk but it still won't open the normal camera application. (sorry for using translation)

JesusSavesBro, 2023-06-15:
I flashed the zip successfully but couldn't find the app

a406010503, 2023-06-18:
You definitely haven't read the instructions carefully, if you have, you won't have this problem.
The problem is that you haven't done the Core Patch program, which is necessary to avoid application loss on the A13.

closed 2023-06-18 by a406010503

JesusSavesBro, 2023-06-19:
> You definitely haven't read the instructions carefully, if you have, you won't have this problem. The problem is that you haven't done the Core Patch program, which is necessary to avoid application loss on the A13.

oh alright please what do i have to do to get the Core Patch program done?